mmccoyd/hillside #1: Slave-side key presses are not registered by master side

jurgen-kluft, 2022-03-29:
Current situation:
Left and right board soldered diodes + reset + TRRS.
Both sides have a Pro Micro flashed with default firmware (cooked by 'https://config.qmk.fm')
Both the left and right side work by themselves, all keys are working.

When testing for continuity (USB disconnected) and TRRS connected, the 'data' pin of the left Pro Micro is connected to the right Pro Micro 'data' pin. VSS and GND are also showing continuity.

So currently I suspect the firmware to not recognize that it should act as 'slave'.

Any suggestions/pointers are very much appreciated at this point.

jurgen-kluft, 2022-03-29:
For my master/slave boards with Pro Micro I had to add the following define into config.h:
#define SPLIT_USB_DETECT

This solved the issue of the master not receiving slave key presses since the slave side did not recognize itself correctly as slave.

mmccoyd, 2022-03-29:
From your other issue #2, it seems you are running a much earlier version of QMK. The split is defined in info.json, which is likely not read correctly by a much earlier version of QMK.

mmccoyd, 2022-03-29:
I'll create a test keymap with a keymap.c in the morning. The default keymap.json creates a simple keymap.c in the .build directory, but an explicit test and example is better.

jurgen-kluft, 2022-03-29:
Same here, will do a run with and without that define to see if the issue is reproducible now with the new QMK firmware.

jurgen-kluft, 2022-03-30:
I have removed/added my fork as well as my installation of QMK so I am now on the latest version `QMK Firmware 0.16.7`.

I tested as follows:
1) Compile my custom keymap for Hillside, test left-side and right-side
2) Add `#define SPLIT_USB_DETECT` to my config.h, compile and test left/right side

Only with 2) am I able to get key presses registered from the slave side.

Note: you can find my hillside keymap here: github.com/jurgen-kluft/qmk-keyboards/hillside

mmccoyd, 2022-03-30:
Your keymap is a _tad_ complex, though worthy of further study. I believe SPLIT_USB_DETECT is old and no longer needed. Turning it on or off did not seem to affect the below keymap.

Here is a simple .c based default that I'll create a PR for in a bit. It builds cleanly with 0.16.7. It includes a commented out line in config.h to set MASTER_RIGHT. I have it in 0_1/keymaps/, but it seems to work fine in keymaps/. As is it works fine with left connected. With the MASTER_RIGHT uncommented it works fine with right connected. 
[default_dot_c.zip](https://github.com/mmccoyd/hillside/files/8376784/default_dot_c.zip)

Let me know if this default_doc_c works for you. What stood out to me in yours was the KYRIA_TO_HILLSIDE_LAYOUT mapping.

(My in-process design avoids this left right nonsense with a diode. Something I need to change on this PCB.)

jurgen-kluft, 2022-03-30:
From the conversation I had with Drasha, he mentions this "because a lot of the newer pro micro clones don't properly detect vbus levels". That means that indeed the slave will presume that it is master and that is exactly what I am seeing. So it is a hardware dependency, if you have these kind of Pro Micros you might need to define SPLIT_USB_DETECT.

https://docs.qmk.fm/#/feature_split_keyboard?id=hardware-configuration-options

In what way did this stand out to you? `What stood out to me in yours was the KYRIA_TO_HILLSIDE_LAYOUT mapping.`

mmccoyd, 2022-03-30:
Drasha knows far far more than I do, but the board has fixed master behavior. It _is_ master left, as I understand and designed, unless it is told in config to _be_ right. So I'm not sure how a clone hardware issue could confuse that.

On the mapping, just that that looked like an extra layer of complexity that would be easy to get wrong. Though unlikely to make one side not work. So not really an issue for this problem.

mmccoyd, 2022-03-30:
Ahh, well it is fixed based on config defaults for the board. If the keymap changes those defaults, then that no longer holds.

jurgen-kluft, 2022-03-30:
Yes I understand that, but to get back to basics, when I use the default hillside keymap my slave side doesn't work. Then adding SPLIT_USB_DETECT to the default keymap fixes it.

You can clearly see in the QMK code that SPLIT_USB_DETECT will identify master by actual USB activity and not by reading vbus.

mmccoyd, 2022-03-30:
Oh,  I thought it _did_ work with the default, but not with your keymap.

What about with the default_dot_c keymap folder, if you set MASTER_RIGHT in config.h?
And alternatively explicitly MASTER_LEFT?

If it is a hardware issue, I'm not sure of a solution beyond explicit config, or new hardware.

mmccoyd, 2022-03-30:
I'm assuming the code and hardware should be able to distinguish between noise vs data in from USB or TRRS and decide based on fixed master assignment what it should then do.

mmccoyd, 2022-03-30:
Did not mean to close it, fumble fingered.

jurgen-kluft, 2022-03-30:
I am sure this is not a hardware issue, for me the only suggestion I can give at this moment is for you to add a note in your documentation about this so that people using Pro Micro MCU's can avoid this specific issue.

I will test the zip that you posted and use MASTER_RIGHT although I am sure that this also will work. However this makes it more involved and risky as a user to flash the firmware correctly since you have to do a 'compile/flash/change/compile/flash'.

In your documentation I suggest you add the comment to enable/add SPLIT_USB_DETECT to their own config.h when they use Pro Micro MCU's.

mmccoyd, 2022-03-30:
Huh. You flash the same firmware to both sides. You do have to pick which side you want the cable to go on though. 

I'll get a pair of Pro-micros to test. But you mentioned _bad_ clones. If the issue is bad clones, that is separate. 

In any case, I'll add some trouble shooting notes. Thanks for the feedback and sorry it did not work quite right with your boards.

jurgen-kluft, 2022-03-30:
Thanks Michael, to be honest the board is great, like the minimal feel and the low profile. So I should be the one thanking you for putting in the design work, the detailed documentation and making it open-source.

Thank you !

Note: Drashna directed me to read this article if I was interested in the details (and trouble) on how the split code is determining master/slave. 
https://medium.com/@keebio/the-case-of-the-wayward-elite-c-73f0fd691f88#3bf5

mmccoyd, 2022-03-30:
Thanks! Creating it scratched an itch and having other people actually use it is really awesome. I have a version in the works that uses a diode in the key matrix on one side to know which side is which, and which has a slight ring and pinky finger splay. Once that is out I'll update this version with the same handedness detection. Hopefully one of all these works out for you. 

I'll read up on that article. Thanks. I do remember a while ago reading about voltage levels detection and what not.

closed 2022-04-04 by jurgen-kluft