mmccoyd/hillside #2: QMK: compiling directly with 'qmk compile' gives errors

jurgen-kluft, 2022-03-29:
[qmk_compile_hillside_errors.log](https://github.com/mmccoyd/hillside/files/8369077/qmk_compile_hillside_errors.log)

These errors can be fixed by adding the following into config.h, however this does have qmk spew out warnings.
#define VENDOR_ID       0xFEED
#define PRODUCT_ID      0x67C0
#define DEVICE_VER      0x0001
#define DIODE_DIRECTION COL2ROW
#define RGBLED_NUM 5
#define RGB_DI_PIN D3
#define SOFT_SERIAL_PIN D2

mmccoyd, 2022-03-29:
Jurgen, Thanks for the report. 

It looks as if you are running a much older version of QMK. Your error log lists QMK Firmware 0.12.50, while  0.16.0 started in February. Recent changes have more more config into info.json, which is likely not read correctly by prior versions.

jurgen-kluft, 2022-03-29:
Hi @mmccoyd,

Yes you are right, after upgrading QMK (correctly this time) it does indeed not complain anymore.

closed 2022-03-29 by jurgen-kluft