mmccoyd/hillside #5: Build guide doesn't include information about wireless builds

rmwphd, 2022-04-12:
The build guide for the Hillside is fantastic, but it's missing a page describing how to hook up the battery, off switch, and capacitor. Which set of batt pos / gnd do you solder the battery leads to?

It's also missing info about closing the jumper for i2c things on the page on adding the resistors.

rmwphd, 2022-04-16:
For those seeking info: The capacitor solders on near the top of the MCU area. Soldering the battery leads to the top pads on the PCBs, near the USB-C ports on your nice!nano, works. The on-off switch goes on just like you imagine it does. 

I don't have the Pimoroni trackball working in ZMK on my board yet, so I don't know if I did the right thing by closing the jumpers on both halves for i2c yet.

mmccoyd, 2022-05-27:
Sorry I missed this. I added a wireless page to the build guide. And a page explaining the accessory power jumper.

closed 2022-05-27 by mmccoyd