mmccoyd/hillside #9: Only USB-C ports

Lumi317, 2022-10-07:
Would it be possible to create variations of the Hillsides whose TRRS ports are replaced by USB-C ports?

mmccoyd, 2022-11-06:
I continue to think about USB, but I'm not there yet. There are a few factors:

- Pro: 
  - It solves worry about frying things when un/plugging the two sides.
  - PCBA is easy enough to set up as I use it for other parts of the boards.
- Con:
  - USB-C forces the more expensive PCBA over hand soldering for everyone. The drag soldering needed for USB-C  is unlikely to be at all feasible those new to soldering or with less optimal equipment or solder.
  - Managing two versions of four boards, for a TRRS or USB solution to that, is non trivial.
  - Micro USB is a possible option that:
     - may still be hand solderable or can share a TRRS footprint.
     - has a few cables available, but not that many, and may fade to history longer term.
  - The MCU must be rewired, likely to RAW instead of VCC. A minor issue.

I may make the switch, as bare boards are not that much cheeper than PCBA, but some other improvements are higher on the list. But thanks.

Lumi571, 2022-11-08:
From my point of view, the USB-C version should just replace the TRRS version rather than maintaining two versions. I think people who are into soldering by hand wouldn't mind giving up manually soldering a TRRS port if it brings the qualities of USB-C to their hardware.
Here is a discussion about it on Reddit that might interest you: https://www.reddit.com/r/ErgoMechKeyboards/comments/rt083u/as_a_reminder_while_the_usb_is_plugged_do_not/

mmccoyd, 2022-11-08:
A good discussion, though not anything new. As Thomas says, debate for relacing it has been ongoing.

mmccoyd, 2022-11-08:
A new addition to the factors list:

- Preventing phone plugged into split connect negotiating higher voltage charge from desktop through keyboard and thus frying keyboards MCU. Should just be by not passing those negotiation circuits to the split connection.