mmccoyd/hillside #13: jlcpcb has issues reading gerber files for 52 & 56

tsteffek, 2022-10-30:
I just wanted jlcpcb to make me some hillside52 PCBs, when I realized that they don't produce sensible prices. I checked the gerber view, and while it does present you with the board in the preview, the logging on the side tells you:
```
Analysis Results
layers : -1
minimum trace width : >=10 mil
minimum trace spacing : >=10 mil
minimum drill size : null
width : null
height : null
Unable to identify the layers due to the non-standard filename
Can not identify the board outline
...
```
I've tried with 56 for the same result and compared those to 46 and 48, which both seem to work (gets correct pricing, and Gerber Viewer doesn't complain about anything).

Disclaimer: This is my first time ordering PCBs and I have no idea what I'm doing, so while it's completely possible I messed up somewhere, I always thought downloading and uploading a .zip was fool-proof :D

mmccoyd, 2022-10-30:
Sorry about that. I get the same result, which should not be the case. 
I'll shift to making them the same way I make the 46 files, which will give you a PCBA option as well if you want it.
That might take a day or so.
Thanks for the issue report!

tsteffek, 2022-10-31:
That would be more than I ever dreamt of! Thank you.

mmccoyd, 2022-11-03:
I pushed a fix for this to https://github.com/mmccoyd/hillside/releases. 

The wiki is updated on how to order them, it is now the same as for the 46. 

Please let me know if that works fine for you! I've tested it with the JLC PCBA parts selection and preview steps for both front and back PCBA.

mmccoyd, 2022-11-03:
I fixed it for the 52. Not yet for the 56...

mmccoyd, 2022-11-08:
I'm fixing this for 56, which will also enable PCBA for it and put its Gerber and PCBA files on the releases page. Though its firmware will still be on a fork, not yet in QMK main.