mmccoyd/hillside #16: Add more parts to PCBA

auipga, 2022-11-26:
Hi,
first of all, thank you for this sophisticated piece of awesomeness! Even the LEDs are hidden _behind_ a key. You're a genius!

I've read the Wiki and if I'm right I need to solder some parts myself after PCBA service:
- SK6812MINI-E available at https://jlcpcb.com/partdetail/OpscoOptoelectronics-SK6812MINIE/C5149201
- Machine pin female sockets 
- Mill-Max Hot Swap Sockets like these https://splitkb.com/products/mill-max-hot-swap-sockets?variant=39472159064141 (IIRC they should fit)

Is it possible to add those to the PCBA too?
Regards

mmccoyd, 2022-11-26:
Hi. Thanks.

A short reply for the moment.

The LEDs could be possible, but they are only supported for the Standard PCBA process, not the Economical one. Which raised the setup price from a modest $8 per side to $25 per side. They are one of the more challenging bits to solder, so it might be good idea, if I can prevent it causing confusion for the normal Economical PCBA process. Thanks.

A risk of adding the encoders to the PCBA is that people who did not really want encoders might still just send the files as is without checking them and deselecting the encoders. I'd say encoders are as easy to solder as switches, and more of a pain to desolder.

A different PCBA manufacturer provides more capabilities, but at higher price. I've not yet looked at using them. I expect they could do the MCU sockets, but I'm not sure about the switch Mil-Max.

auipga, 2022-11-27:
Hi Michael. Thank you for your detailed reply.

The LEDs would be the most interesting. What I didn't understand is whether the more expensive standard PCBA Type is evitable or not.

The rest would be ok to solder myself.

auipga, 2022-11-30:
Now that I decided to pay that price for the standard PCBA type, I'd like to add whatever is possible.
Could you help me achieve this / guide me through the process?
All I can get to is this:
[image]

What is the correct way to add and position new parts to existing footprints?

mmccoyd, 2022-11-30:
Hi. I'm adding that to the release zip, thought it might be a day or two. 

Either of the power switches should be fine. Moving to oval holes expanded the switch choices, but I've not fully updated the docs yet.

auipga, 2023-01-09:
Hi, just wanted to ask whether I can do something for you to get it done?

closed 2023-02-22 by auipga