mmccoyd/hillside #17: [Somebody interested?] Bigger battery on board

auipga, 2022-11-26:
It's just an idea, not yet as sophisticated as your work ;)
But fixing an empty battery within 5 seconds replacing it, putting the empty one into the charger and use it for weeks/months before you need to repeat? Sounds good to me.

https://www.amazon.de/-18650-battery-holder/dp/B081HW4TGR

Pinta crashed while painting mockup :'( So here's the text-only-edition:
I'd place the battery right above the first thumb key with enough space for an OLED ;)
What do you think?

[image]

johannes-jp, 2024-12-06:
Hey I did this! With my placement it fills the space underneath the palm, and stays out of the way of the thumb. I didn't want it on the inside between the boards, because I prefer to use that space for a mouse. 
[image]
[image]
It doesn't look very nice and I made a 3d model for doing it properly, which I can upload when I find it.

auipga, 2024-12-21:
Looks promising :) thanks for sharing

tokazio, 2024-12-31:
I've used 3000mAh flat one.
At the top back that give some tenting 😉[image]

[image]

[image]

(The keyboard is a sofle choc but i've done the same thing on my dailly hillside)