mmccoyd/hillside #19: right half does not work with stock (52) zmk firmware

ramnathk, 2023-01-01:
Hi there I just finished assembling my hillside 52. The right half does not work when I flash it with the right nice nano firmware. However, if I flash it with the left's firmware, it works. What is amiss? As I understand I can leave the handedness jumpers open as ZMK handles it in software. So, what's the issue?

I use the QMK configurator's keyboard tester to test for keystrokes
Cheers,
Karthik

autoferrit, 2023-05-22:
i am planning on ordering the 52 to use with zmk and nice nanos. did you ever figure this out?

auipga, 2023-05-31:
@ramnathk I'd guess, that you tried mmccoyd's default firmware and settings_reset. You could provide your firmware.zip and I'll try it on mine, just to ensure it's a hardware situation. I would also try to switch the controllers in case you used the socket option.
@autoferrit mine works, so it's not a general problem.

autoferrit, 2023-06-01:
got it. i did have an issue with it working with the nice nanos after the recent zephyr update. but i managed to update the `config/boards/shields/hillside52/nice_nano_v2.overlay` and got it to compile properly. so it seems to be working so far. though I just did this and might need to be tested more.

tokazio, 2023-10-13:
@autoferrit i have the same problem, what have you done to make the right part work please ?