mmccoyd/hillside #20: How to mount encoder

tim-hilt, 2023-03-13:
I have ordered all required parts to build my Hillside. The only thing I don’t really know with is how to solder the encoders. One of the holes in the PCB doesn’t seem to be wide enough.

Here’s a photo of what I mean:

[image]

How am I supposed to deal with this? Do I just cut off the second wide leg?

mmccoyd, 2023-03-14:
Tim, Yes, just cut off the circled wide leg. I'd update the docs if it's not there. A wide leg hole and LED would not go well together. I hope you like the board.

tim-hilt, 2023-03-14:
Thanks for the response! :) I’m very excited to see how the Hillside works for me! Thank you!

closed 2023-03-14 by tim-hilt