mmccoyd/hillside #34: Rotations slightly off for diodes in bottom placement file

dalewking, 2024-08-22:
The rotation angle is a bit off for the diodes in the bottom side placement file when those are at a slight angle.

AThilenius, 2026-01-02:
@dalewking Do you know if JLC is able to fix that during review? I want to throw in an order, but lack the knowledge with KiCad to fix them 😬

dalewking, 2026-01-04:
It's been a while, but I was able to fix it when submitting the order, but it wasn't automatic I had to manually adjust the rotation manually for each of the misplaced diodes.

I have some unused boards since I had to order 5 sets and could sell you a set if it makes sense for the cost of shipping.

I will also add that I have tried so many times to get used to this keyboard, but the biggest problem is the really aggressive pinky stagger. I am contemplating trying a different keyboard or trying to design something new closer to Corne or Lily 58. Ideally want something with a track pad and rotary encoders. I don't really think the nice nano display is that necessary

dalewking, 2026-01-07:
> I will also add that I have tried so many times to get used to this keyboard, but the biggest problem is the really aggressive pinky stagger.

I am totally changing my mind on this. I was using the rather flat [MBK style of low profile Choc keycaps](https://typeractive.xyz/products/mbk-keycaps?srsltid=AfmBOooMI-8FWhbkm4w5_ejC4cW7kFdZneeb9Y2gdvV1Zl6IHf90-MRp) and it was a struggle to keep my fingers in the right position and was constantly hitting the wrong key.

I have switched to [KLP Lame 3d printed keycaps](https://www.printables.com/model/552557-klp-lame-kailh-low-profile-choc-keycaps/related) and it is a night and day difference and am now happily typing on my  HSV effortlessly!